sifive/sifive-blocks #83: chiplink causing compile errors with chisel in latest rocket-chip

juliusbaxter, 2018-08-28:
I've got a project based on the Freedom design. I've just rebased our stuff on top of the latest in rocket-chip's master branch and now see a bunch of issues with chiplink, it primarily seems to do with things which are no longer present in Chisel.Bundle:

```
[info] Compiling 63 Scala sources to /work/sifive-blocks/target/scala-2.12/classes ...
[error] /work/sifive-blocks/src/main/scala/devices/chiplink/CAM.scala:24:6: value key is not a member of Chisel.Bundle
[error]   io.key := OHToUInt(free_sel, keys)
[error]      ^
[error] /work/sifive-blocks/src/main/scala/devices/chiplink/CAM.scala:26:6: value alloc is not a member of Chisel.Bundle
[error]   io.alloc.ready := free.orR
[error]      ^
[error] /work/sifive-blocks/src/main/scala/devices/chiplink/CAM.scala:27:12: value alloc is not a member of Chisel.Bundle
[error]   when (io.alloc.fire()) { data.write(io.key, io.alloc.bits) }
[error]            ^
[error] /work/sifive-blocks/src/main/scala/devices/chiplink/CAM.scala:27:42: value key is not a member of Chisel.Bundle
[error]   when (io.alloc.fire()) { data.write(io.key, io.alloc.bits) }
[error]                                          ^
[error] /work/sifive-blocks/src/main/scala/devices/chiplink/CAM.scala:27:50: value alloc is not a member of Chisel.Bundle
[error]   when (io.alloc.fire()) { data.write(io.key, io.alloc.bits) }
[error]                                                  ^
[error] /work/sifive-blocks/src/main/scala/devices/chiplink/CAM.scala:30:19: value alloc is not a member of Chisel.Bundle
[error]   val bypass = io.alloc.fire() && io.free.bits === io.key
[error]                   ^
[error] /work/sifive-blocks/src/main/scala/devices/chiplink/CAM.scala:30:38: value free is not a member of Chisel.Bundle
[error]   val bypass = io.alloc.fire() && io.free.bits === io.key
[error]                                      ^
[error] /work/sifive-blocks/src/main/scala/devices/chiplink/CAM.scala:30:55: value key is not a member of Chisel.Bundle
[error]   val bypass = io.alloc.fire() && io.free.bits === io.key
[error]                                                       ^
[error] /work/sifive-blocks/src/main/scala/devices/chiplink/CAM.scala:31:29: value alloc is not a member of Chisel.Bundle
[error]   io.data := Mux(bypass, io.alloc.bits, data(io.free.bits))
[error]                             ^
[error] /work/sifive-blocks/src/main/scala/devices/chiplink/CAM.scala:31:49: value free is not a member of Chisel.Bundle
[error]   io.data := Mux(bypass, io.alloc.bits, data(io.free.bits))
[error]                                                 ^
[error] /work/sifive-blocks/src/main/scala/devices/chiplink/CAM.scala:34:20: value alloc is not a member of Chisel.Bundle
[error]   val clr = Mux(io.alloc.fire(), free_sel, UInt(0))
[error]                    ^
[error] /work/sifive-blocks/src/main/scala/devices/chiplink/CAM.scala:35:20: value free is not a member of Chisel.Bundle
[error]   val set = Mux(io.free.valid, UIntToOH(io.free.bits), UInt(0))
[error]                    ^
[error] /work/sifive-blocks/src/main/scala/devices/chiplink/CAM.scala:35:44: value free is not a member of Chisel.Bundle
[error]   val set = Mux(io.free.valid, UIntToOH(io.free.bits), UInt(0))
[error]                                            ^
[error] /work/sifive-blocks/src/main/scala/devices/chiplink/ChipLink.scala:169:11: value b2c_data is not a member of Chisel.Bundle
[error]     rx.io.b2c_data := port.b2c.data
[error]           ^
[error] /work/sifive-blocks/src/main/scala/devices/chiplink/ChipLink.scala:170:11: value b2c_send is not a member of Chisel.Bundle
[error]     rx.io.b2c_send := port.b2c.send
[error]           ^
[error] /work/sifive-blocks/src/main/scala/devices/chiplink/ChipLink.scala:171:25: value a is not a member of Chisel.Bundle
[error]     out.a <> sourceA.io.a
[error]                         ^
[error] /work/sifive-blocks/src/main/scala/devices/chiplink/ChipLink.scala:172:25: value b is not a member of Chisel.Bundle
[error]     in .b <> sourceB.io.b
[error]                         ^
[error] /work/sifive-blocks/src/main/scala/devices/chiplink/ChipLink.scala:173:25: value c is not a member of Chisel.Bundle
[error]     out.c <> sourceC.io.c
...
[error] /work/sifive-blocks/src/main/scala/devices/chiplink/Partial.scala:30:25: value === is not a member of Any
[error]   val enable = i_opcode === TLMessages.PutPartialData
[error]                         ^
[error] /work/sifive-blocks/src/main/scala/devices/chiplink/Partial.scala:34:32: value << is not a member of Any
[error]     val wide = shift | (i_data << (state << 2))
[error]                                ^
[error] /work/sifive-blocks/src/main/scala/devices/chiplink/Partial.scala:35:12: value := is not a member of Any
[error]   Expression does not convert to assignment because receiver is not assignable.
[error]     o_data := Vec.tabulate(4) { i => wide(9*(i+1)-1, 9*i+1) } .asUInt
[error]            ^
[error] /work/sifive-blocks/src/main/scala/devices/chiplink/Partial.scala:36:12: value := is not a member of Any
[error]   Expression does not convert to assignment because receiver is not assignable.
[error]     o_mask := Vec.tabulate(4) { i => wide(9*i) } .asUInt
[error]            ^
[error] /work/sifive-blocks/src/main/scala/devices/chiplink/Partial.scala:40:10: value i is not a member of Chisel.Bundle
[error]       io.i.ready := Bool(true)
[error]          ^
[error] /work/sifive-blocks/src/main/scala/devices/chiplink/Partial.scala:41:10: value o is not a member of Chisel.Bundle
[error]       io.o.valid := Bool(false)
[error]          ^
[error] /work/sifive-blocks/src/main/scala/devices/chiplink/Partial.scala:45:14: value i is not a member of Chisel.Bundle
[error]     when (io.i.fire()) {
[error]              ^
[error] /work/sifive-blocks/src/main/scala/devices/chiplink/Partial.scala:46:27: type mismatch;
[error]  found   : Any
[error]  required: chisel3.core.Data
[error]       shift := Mux(empty, i_data, wide >> 36)
[error]                           ^
```
I was going to simply remove chiplink from anywhere it's referenced in my design for now, but I'm at a loss as to why it's compiling it in the first place. It's not imported anywhere, and there's no reference to it outside of its directory in. 

¯\\\_(ツ)\_/¯

juliusbaxter, 2018-08-28:
Actually if I remove the chiplink directory it appears the same thing happens with the next block it picks up.

I'm wondering if it's something in my setup, things look very broken :(

Does anyone have any suggestions?

jackkoenig, 2018-08-28:
This is actually due to one of the very few backwards incompatible changes between Scala 2.11 and 2.12. This can be resolved by including the scalac option `-Xsource:2.11` in your build. See the [rocket-chip build.sbt file](https://github.com/freechipsproject/rocket-chip/blob/25b5a25b75d759eae2d7dad95c8729cd07628e7b/build.sbt#L17) for an example:

juliusbaxter, 2018-08-29:
Awesome, thanks again @jackkoenig that did the trick.

closed 2018-08-29 by juliusbaxter