sifive/sifive-blocks #101: How to modify periphery to use other clock and cross clock

colin4124, 2019-01-04:
Take GPIO for example,  there are `controlXType` and `mclock` in `GPIOAttachParams` .

https://github.com/sifive/sifive-blocks/blob/f79faa262dfad74d81d9bffd5469ad3de6fb244e/src/main/scala/devices/gpio/GPIO.scala#L215-L223

The code below is to use it:

https://github.com/sifive/sifive-blocks/blob/f79faa262dfad74d81d9bffd5469ad3de6fb244e/src/main/scala/devices/gpio/GPIOPeriphery.scala#L10-L12

If I want to use other clock giving to GPIO, should I use `SynchronousCrossing` ?  So I would give `controlXType` as below:

```scala
trait HasPeripheryGPIO { this: BaseSubsystem =>
   val gpioXType = SynchronousCrossing // Here I use dafault options
   val gpioNodes = p(PeripheryGPIOKey).map { ps => GPIO.attach(GPIOAttachParams(ps, pbus, ibus.fromAsync, controlXType=gpioXType)).ioNode.makeSink } 
} 
```
**But I don't know how to connect `mclock` from outside, such as GPIO's IO port.**

colin4124, 2019-01-04:
@borancar Would you have any idea? PS: I think github issues is much useful than sifive forums, because I can link the code and jump to source file easy.

borancar, 2019-01-04:
Depends what the relationship between the clocks will be. If the clocks have different frequencies and phases then you typically need AsynchronousCrossing. As for the mclock and mreset, you just need to pass it to the GPIOAttachParams and the magic will happen in attach:

https://github.com/sifive/sifive-blocks/blob/f79faa262dfad74d81d9bffd5469ad3de6fb244e/src/main/scala/devices/gpio/GPIO.scala#L239-L240

This is just what I think should be done, and Diplomacy will do its magic. Note that I haven't tried any of it yet.

colin4124, 2019-01-07:
**I use different way to solve problem.**

1. delete the "Specify module clock" in attach method:
https://github.com/sifive/sifive-blocks/blob/f79faa262dfad74d81d9bffd5469ad3de6fb244e/src/main/scala/devices/gpio/GPIO.scala#L239-L240

2. Add explicitly **Clock** and **Reset** to `GPIOPortIO`

```scala
class GPIOPortIO(val c: GPIOParams) extends Bundle {
  val new_clock = Input(Clock())
  val new_reset = Input(Bool())
  val pins = Vec(c.width, new EnhancedPin())
  val iof_0 = if (c.includeIOF) Some(Vec(c.width, new IOFPin).flip) else None
  val iof_1 = if (c.includeIOF) Some(Vec(c.width, new IOFPin).flip) else None
}
```

3. Use `LazyRawModuleImp` and `withClockAndReset` instead of `LazyModuleImp` and implicitly `clock` and `reset`:

```scala
lazy val module = new LazyRawModuleImp(this) {
   withClockAndReset(port.new_clock, port.new_reset) {
      .....
   }
}
````