sifive/sifive-blocks #154: problem:  the code generated by rocket-chip with i2c

q799807612, 2020-07-09:
i  use the latest rocket-chip and i2c in sifive-blocks to generate SOC,but i2c's output is directly deasserted  in  the verilog code generated by rocket-chip，as follows ：
[image]
i can't use i2c driver program to output. so , i wonder whether  i make some mistakes or i2c in sifive-blocks can't be used.

mwachs5, 2020-07-09:
Which method are you using to instantiate the i2c?

zhvn100, 2020-07-09:
i  add traits like this and add params as follows:
[image]
[image]
gpio、uart、spi can be used when i run program ,but i2c's output  is deasserted in verilog

tamood, 2020-12-10:
Well, the implementation of I2C comes from opencores implementation of I2C, which is not the best but it works with proper electronics. The short answer is as follows.
1) Plug I2C scl and sda ports into IOBUF, if implementing in Xilinx, followed by pulling up the pins with 1kohm resistances.
2) If you are interfacing with a standard I2C bus like one in Arduino, an external pull-up is not required.

Hope it may help, for further reference, see the code at

https://github.com/tamood/freedom/tree/u1-tamood