sifive/sifive-blocks #173: gpio connection with Rocket Chip

Uzair540, 2020-12-24:
I want to connect gpio with my rocket chip can you eleborate me which module of verilog I have to target ? These are the module list of geberated verilog of rocket chip.
module AXI4Buffer
module AXI4UserYanker
module AXI4Deinterleaver
module AXI4IdIndexer
module TLToAXI4
module AXI4UserYanker_1
module AXI4Fragmenter
module AXI4IdIndexer_1
module AXI4UserYanker_2
module AXI4IdIndexer_2
module AXI4RAM
module AXI4Xbar
module AXI4Buffer_1
module AXI4Fragmenter_1
module AXI4RAM_1
module AXI4Xbar_1
module AXI4Buffer_2
module AXI4Fragmenter_2

tamood, 2020-12-28:
@Uzair540 Do not look into emitted Verilog listings. GPIO is connected to rocket-chip with HasPeripheralGPIO and HasPeripheralGPIOImp mix-ins.

Uzair540, 2020-12-28:
@tamood can you elaborate me should I have to write the trait of HasPeripheralGPIO or it is by default available ?If available kindly guide me the path

Uzair540, 2020-12-28:
@tamood I have attached gpio block from sifive to my rocket chip.Can you help me how can I connect my gpio with rocket chip.As When I connect gpio and run rocket chip it is giving me I/O errors as I haven't connect them in code Just below
[image]
I have do the following steps 
In ExampleRocketSystem I have made the following changes

[image]

And in Config subsystem file I made the following changes

[image]

If you help me How can I connect gpio So that my I/O reduces