sifive/sifive-blocks #175: Chiplink Assertion Failed

retrhelo, 2021-11-17:
I am running some post-simulations with a SoC using Chiplink. However, I catch this error message that terminates the simulation without timing violation mentioned. 

> Assertion failed: 'D' channel acknowledged for nothing inflight (connected at ChiplinkTest.scala: 220:11)
>    at Monitor.scala:467 assert((a_set | inflight)(bundle.d.bits.source), "'D' channel acknowledged for nothing inflight" + extra)
> Fatal: "../dev/chiplink/top.v", 76658: asic_system.u0_Chiplink_DualTop.xbar_1.dual_TLMonitor: at time 165740000 ps
> $finish called from file "../dev/chiplink/top.v", line 76658.

I'm not familiar with Chiplink protocol, and there seems very few documentations about it. I wonder what this fatal messages means.

retrhelo, 2021-11-17:
By the way, is there any existing document that's able to give a relatively detailed description about ChipLink Protocol? I spent lots of time searching the Internet, but only to find things about TileLink... :/

mjolarpet, 2021-11-17:
@kritikbhimani

kritikbhimani, 2021-11-17:
First basic check is to look into reset and clock supplied to chiplink block and the rest of the soc. 
Chiplink is mentioned here https://sifive.cdn.prismic.io/sifive/d3ed5cd0-6e74-46b2-a12d-72b06706513e_fu540-c000-manual-v1p4.pdf and here https://www.crowdsupply.com/microchip/hifive-unleashed-expansion-board
I can help solve any of your queries in understanding the implementation of chiplink in scala/chisel found here https://github.com/sifive/sifive-blocks/tree/master/src/main/scala/devices/chiplink

retrhelo, 2021-11-17:
Thank you very much! So if I understand it correctly, chiplink is basically the same as TileLink in timing behavior, but implemented off-chip?

kritikbhimani, 2021-11-17:
Yes https://forums.sifive.com/t/chiplink-isnt-an-open-spec-apparently-or-can-i-see-it-somewhere/1479

retrhelo, 2021-11-17:
Thanks again. My team thinks that it is some unknown timing violations that trigger this assertion. Knowning the RTL timing behavior about ChipLink will help us a lot locating the violation. :D

kritikbhimani, 2021-11-17:
Timing violations come into picture post SYNTHESIS (TLMonitor should not be synthesized)

retrhelo, 2021-11-17:
Yes, we're connecting the SoC with external FPGA using a ChipLinkBridge. And the FPGA part is a simulated one in the post-simulation work flow.

closed 2021-11-18 by retrhelo