sifive/sifive-blocks #178: Arty100TShell - Connecting GPIOPMOD

bchetwynd, 2022-06-07:
The Arty100T shell coupled with the GPIOPMODOverlay.scala provides methods for connecting GPIO signals via one of the PMOD connectors, but it is unclear how to connect the top-level IO in the test harness to the overlay.

For example, here are the methods for mapping the UART I/O:
```
val io_uart_bb  = BundleBridgeSource(() => (new UARTPortIO(dp(PeripheryUARTKey).head)))
dp(UARTOverlayKey).head.place(UARTDesignInput(io_uart_bb))
```

A similar method does not exist for the GPIOPMODDesignInput since it takes no parameters (vs a BundleBridgeSource as seen in other periphery).

This may be somewhat deliberate since the GPIO Sifive-Block uses the GPIOPortIO bundle type vs the overlay using the GPIOPMODPortIO bundle type.

Is there guidance/examples on how to make this connection?

Usutatsu, 2022-08-24:
I want to be able to connect my new Pmod module to my FPGA board. 
The PMOD module to be connected must be connected via SPI interface.
So I want to connect the new SPI interface with a vacant PMOD connector.
Does anyone know which part of the description I need to change to create the new SPI interface?
I would be very happy if you could tell me.

bchetwynd, 2022-08-30:
I'd recommend taking a look @ https://github.com/ucb-bar/chipyard/tree/arty100t

This has a PMOD-based SPI interface.

closed 2022-08-30 by bchetwynd